lukaskremla/micropython-tools-jetbrains #1: Settings indefinate load

jaylinwylie, 2025-01-15:
### Issue
The settings page of the plugin fails to load and displays a spinning wheel indefinitely.

### Environment
- **IDE Version:** PyCharm 2024.3.1.1, Professional Edition
- **Operating System:** Linux (Ubuntu GNOME)

lukaskremla, 2025-01-15:
> ### Issue
> The settings page of the plugin fails to load and displays a spinning wheel indefinitely.
> 
> ### Environment
> * **IDE Version:** PyCharm 2024.3.1.1, Professional Edition
> * **Operating System:** Linux (Ubuntu GNOME)

Could you please try to fully uninstall the plugin, [invalidate all caches](https://www.jetbrains.com/help/idea/invalidate-caches.html), reinstall the latest 0.2.1 version from the plugin marketplace and retry in a newly created project?

jaylinwylie, 2025-01-15:
No luck after invalidating cache completely.
On a new project, the setting page _would_ load until I enabled the plugin and applied my changes. When attempting to access the setting page from the "No port is selected" popup I receive the following error: 

Connecting... error - IllegalStateException: Cannot find configurable for specified predicate

lukaskremla, 2025-01-16:
> No luck after invalidating cache completely.
>
> On a new project, the setting page _would_ load until I enabled the plugin and applied my changes. When attempting to access the setting page from the [...]

Could you please try again with the new 0.2.2 version?

jaylinwylie, 2025-01-17:
Fixed! the settings page now loads.

closed 2025-01-17 by jaylinwylie