lukaskremla/micropython-tools-jetbrains #3: Can't find any devices on linux?

Nama, 2025-01-28:
Nice to see this plugin after the other one was stopped being maintained \o/

In the settings page and in the bottom windows, I can't select a device.

[image]

Yes, I'm using a ESP8266 D1 mini and you warn about it not being tested. Could be both?

I can successfully use `minicom -D /dev/ttyUSB0`.

Nama, 2025-01-30:
Unchecking the new option lists `/dev/ttyUSB0`, but if I click connected, this appears:

[image]

Nama, 2025-02-04:
Not sure if you really wanna close the issue.
The `Connect error` appears every time and I need to click "Refresh" for it to connect.

lukaskremla, 2025-02-05:
> Not sure if you really wanna close the issue. The `Connect error` appears every time and I need to click "Refresh" for it to connect.

My apologies, I remember experiencing an issue similar to the "IndexOutOfBoundsException" which you described and I took it lightly as that problem went away on it's own, however, I've ran into your problem myself and realized that it's more severe than I thought.

It should be fixed in the latest pre-release version 0.2.5, could you please try this release and tell me if your issue is resolved?

[micropython-tools-jetbrains-0.2.5-pre-release1.zip](https://github.com/user-attachments/files/18671434/micropython-tools-jetbrains-0.2.5-pre-release1.zip)

closed 2025-02-06 by lukaskremla