lukaskremla/micropython-tools-jetbrains #4: WebREPL to ESP32 is not working

rimelis, 2025-02-03:
Hello,

I am trying to connect ESP32 board (version: _MicroPython v1.24.0 on 2024-10-25; Generic ESP32 module with OTA with ESP32_) via WebREPL and immediately receiving such error:

**Connect error - RuntimeExceptionWithAttachments: Access is allowed from Event Dispatch Thread (EDT) only; If you access or modify model on EDT consider wrapping your code in WriteIntentReadAction or ReadAction; see https://jb.gg/ij-platform-threading for details Current thread: Thread[#113,DefaultDispatcher-worker-40,5,main] 1034281838 (EventQueue.isDispatchThread()=false) SystemEventQueueThread: Thread[#47,AWT-EventQueue-0,6,main] 409368297**

Other info:
_PyCharm 2024.3.1 (Community Edition) on Ubuntu 24.04.1 LTS
MicropythonTools v.0.2.3_

lukaskremla, 2025-02-03:
Hello, 

Thank you for the report! I don't use WebREPL myself so I missed this bug completely.

It should be fixed now, the update is pending approval in the plugin marketplace, in the meantime I'm sending you the latest version through here as well. It can be manually installed in the IDE plugin page.

[micropython-tools-jetbrains-0.2.4.zip](https://github.com/user-attachments/files/18648011/micropython-tools-jetbrains-0.2.4.zip)

[image]

lukaskremla, 2025-02-05:
@rimelis The 0.2.4 release seems to have some additional issues. Could you please try this 0.2.5 pre-release and tell me if your problem is resolved?

[micropython-tools-jetbrains-0.2.5-pre-release1.zip](https://github.com/user-attachments/files/18671434/micropython-tools-jetbrains-0.2.5-pre-release1.zip)

lukaskremla, 2025-02-12:
Thank you for getting back to me with the details!

Just to update you about the current situation:

I have been able to replicate your issue and I'm working on a fix. However, the problem is somewhere in the WebREPL websocket communication implementation, which is one of the things that remained unchanged when I started this fork. It's code that I'm not familiar with and I'm not sure how long it will take me to come up with a solution.

Hopefully I'll get back to you with a fix soon.

lukaskremla, 2025-03-23:
Hello,

Sorry for the long delay, I only got to sit down and thoroughly debug the WebREPL process recently. 

Unfortunately, fixing this bug will not be very easy. It appears that the MicroPython WebREPL implementation has a command length limitation. It only accepts commands with a length of up to approximately 500 characters. Longer commands get cut off. I have opened [an issue](https://github.com/micropython/micropython/issues/16997) on the official MicroPython GitHub repository, hopefully I'll be able to get back to you with more details soon.

As for possible fixes, there are two options - Either the limit gets raised in future MicroPython versions or I will have to rewrite how the scripts are executed over WebREPL. The scripts would have to get uploaded to the device's file system and be executed from there.

I hope to avoid the second option. Either way, WebREPL will unfortunately remain unavailable until this gets sorted.

lukaskremla, 2025-03-29:
I have managed to get WebREPL support somewhat working in the new 0.4.2 version. 

It now sends the commands in chunks of no more than 255 bytes and it puts a very generous delay of 200ms between sending out each chunk. This of course significantly reduces the upload speed, so it is strongly recommended to enable FTP uploads if using WebREPL.

This support is still experimental, uploads or command executions can fail unexpectedly. I will try to figure out if the current implementation can be improved and sped up in any way, so upcoming updates should improve the reliability.

Some more information can be found [here](https://github.com/lukaskremla/micropython-tools-jetbrains/blob/0.4.2/docs/v0.4.2/WEBREPL.md).

As of writing this reply the 0.4.2 version which implements this fix is pending approval to be publicly available in the JetBrains plugin marketplace. Until it is approved you can download the archive and manually install it from the [0.4.2 release](https://github.com/lukaskremla/micropython-tools-jetbrains/releases/tag/0.4.2).

More info on how to do this can be found [here](https://www.jetbrains.com/help/mps/managing-plugins.html#install_plugin_from_disk) the option is called "Install Plugin from disk".