lukaskremla/micropython-tools-jetbrains #5: Connection issues with ESP32

Helveg, 2025-02-04:
Thank you so much for this plugin!

I'm using the Arduino Nano ESP32 and your plugin worked flawlessly the first time I used it to flash the project, but ever since it has issues. It doesn't seem to overwrite the file content of existing files (even though I can see an upload progress bar flash by if the file was changed, so I think the diffchecking works), and it gets stuck on "Updating file system view" part when connecting:

antonsterkhov, 2025-02-04:
Connect I/O error - Timeout during command execution:try: import machine import os d = (2025, 1, 4, 22, 33, 41, 0, 0) try: r = machine.RTC() r.datetime(d) except Exception as e: pass b = True s = os.uname() v = s[3] d = s[4] try: import binascii except ImportError: b = False print(v, d, b, sep="&") except Exception as e: print(f"ERROR: {e}")

lukaskremla, 2025-02-05:
Hello,

Thanks for the bug report. I've been able to replicate your issue, but the behavior is very random and inconsistent. It will probably take me a while to sift through the code and test everything to pin-point the issue.

Right now it seems that this issue is present in versions 0.2.3 / 0.2.4. Are you using either one of these versions?

If so, could you please try to manually install version 0.2.2 and tell me if the problems go away?

[micropython-tools-jetbrains-0.2.2.zip](https://github.com/user-attachments/files/18671108/micropython-tools-jetbrains-0.2.2.zip)

[image]

lukaskremla, 2025-02-05:
I've also downgraded some dependencies of the plugin and that seems to have made things stable again. Could you please try the 0.2.2 version and then thins 0.2.5-pre-release1 and tell me if they both work for you?

[micropython-tools-jetbrains-0.2.5-pre-release1.zip](https://github.com/user-attachments/files/18671434/micropython-tools-jetbrains-0.2.5-pre-release1.zip)

And you asked about the plugin getting the board to an unresponsive state, this is most likely caused by the serial access lock. Even other dev tools like Thonny do this - while you're connected through Thonny you can't interact with the board from any other serial communication tool.

This plugin does the same thing, but when its code runs into some errors this behavior can get stuck on locking the device, which would explain the inaccessible state you've described.

If that happens either restarting the device or fully closing out PyCharm should fix it. But it shouldn't happen when the plugin is working the way it's meant to, without running into exceptions and IDE errors.

Helveg, 2025-02-05:
Thanks for the quick response and detailed investigation, your open source commitment is greatly appreciated ❤ 

* I uninstalled my 0.2.4 plugin via the plugin settings, restarted the IDE which then offered the 0.2.2 version on the plugin marketplace (perhaps you downgraded the public version?), and it gives the following error:

```
Connect I/O error - Micropython filesystem scan failed, Traceback (most recent call last): File " ", line 13, in   File " ", line 10, in l File " ", line 7, in l OSError: [Errno 22] EINVAL
```

* Then I uninstalled it, restarted the IDE, and installed your 0.2.2 zip, same error, then I uninstalled it again, restarted again
* Then I installed the 0.2.5 zip and same error again:

```
Connecting... I/O error - Micropython filesystem scan failed, Traceback (most recent call last): File " ", line 13, in   File " ", line 10, in l File " ", line 7, in l OSError: [Errno 22] EINVAL
```

Now I'll be trying out some other things like restarting my laptop, using Arduino Labs etc to see if anything helps me get past the error, and will let you know if I find some meaningful information

Helveg, 2025-02-05:
Hmm, no this Arduino Nano ESP32 is straight out of the box. The only things I've done is use the MicroPython installer, and uploaded some small scripts with Arduino Lab. Arduino Lab still seems to be able to see and manipulate the filesystem:

[image]

antonsterkhov, 2025-02-05:
### I installed 0.2.5 but still get the error
Connect I/O error - Timeout during command execution:try: import machine import os d = (2025, 1, 5, 20, 41, 53, 0, 0) try: r = machine.RTC() r.datetime(d) except Exception as e: pass b = True s = os.uname() v = s[3] d = s[4] try: import binascii except ImportError: b = False print(v, d, b, sep="&") except Exception as e: print(f"ERROR: {e}")

### I have a Mac M1 and Atom Lite

lukaskremla, 2025-02-06:
> Hmm, no this Arduino Nano ESP32 is straight out of the box. The only things I've done is use the MicroPython installer, and uploaded some small scripts with Arduino Lab. Arduino Lab still seems to be able to see and manipulate the filesystem:
> 
> [image]
> 
> Update: I don't have the error anymore, instead I'm back to the `Connect timed out` after "Updating file system view". I'll try v0.2.2 again. --> Nevermind it's all kind of random, I get the EINVAL error usually only the first time after an IDE reboot, followed by timeouts on subsequent attempts. Unplugging the board resets it and makes it usable by Arduino Lab again, but it never works with the MicroPython Tools plugin :( (Arduino Lab is disconnected and closed, so it's not that the port is locked or anything silly)

This is very weird, I'm unable to replicate your issue with my ESP32 boards, they all work with the plugin. It's especially odd seeing as even your board worked the first time you used.

I'm not sure if I can do something to influence the problem with updates from my side, given that it works fine for me and other people as well. 

It seems like the problem might be in your IDE. Could you try to run through the following steps and tell me if the issue went away at the end?

1. Uninstall the plugin fully and restart the IDE
2. [invalidate all caches](https://www.jetbrains.com/help/idea/invalidate-caches.html)
3. Ensure the IDE itself and installed plugins are up-to-date
4. Install the 0.2.5 version of this plugin again
5. Create a new project and try if the issue went away

If neither of those do the trick, it might be worth trying to flash the MicroPython firmware again - I'd recommend doing so through Thonny, as I know that works just fine for me. 

Also, if you have access to some other computer and would be willing to do this - I would try installing PyCharm and this plugin on a different machine and seeing if the behavior is any different. Similarly if you any other microcontrollers at hand, I'd try with them.

If you're willing to go through some more manual steps (and trying on a different computer is not an option), you could try reinstalling PyCharm and/or disabling other plugins in case they are somehow preventing this one from working.

And I know you said that the arduino IDE was turned off when you tried this - but I think this is still worth mentioning. Make sure no other programs could be interfering with this plugin. Did anything change since the plugin work the first time? Maybe different devices connected or different programs running while you try this. 

Sorry for giving you these general suggestions, but right now that's all I can do unless I somehow manage to replicate this issue myself. I really think (and hope) the first five steps I mentioned are what you need.

Helveg, 2025-02-06:
I think the board is malfunctioning. When using `esptool` I get `write timeout` errors from pySerial so I don't think it can be salvaged. I suspect the breadboard I had it in might have shorted some pins or something... Thanks for the help, nothing more to do here, I'll order another board and try again ;)

closed 2025-02-06 by Helveg