lukaskremla/micropython-tools-jetbrains #7: reference library code hint

suecher, 2025-02-22:
I feel that I am indeed more accustomed to using it than Thonny, but I have a question to ask, is it normal for the microPython library prompt in the environment not to support this?

[image]

lukaskremla, 2025-02-22:
If you mean the underlined MicroPython specific library imports in the image, then you need to select and apply a stub package in the settings of the plugin as the requirements inspection "No stub package selected" is instructing.

If I miss understood and your question is somehow related to the way stub package selection is currently implemented - could you please go into more detail?

suecher, 2025-02-22:
It has been configured, thank you very much, the plugin is very easy to use.
 
^0^