lukaskremla/micropython-tools-jetbrains #11: I have a problem with the USB cable and connection to the esp32.

Witawat, 2025-03-05:
Hello Lukas Kremla

I have a problem with the USB cable and connection to the esp32.

If you disconnect the USB cable during the connection without pressing disconnect, 

the program will freeze and cannot be reconnected.

Can you check the connection or fix this problem?

lukaskremla, 2025-03-05:
I'm aware of the plugin not immediately recognizing that the device is no longer connected. However, I've never ran into it causing the IDE to freeze, you should receive some error notification as soon as you do something that would make the plugin try to communicate with the board, but the IDE should remain responsive.

Are you sure it causes the IDE to freeze in a way where it can't recover on its own without you having to restart it?

What IDE are you using? What version is it and what's your operating system?

lukaskremla, 2025-03-07:
@Witawat have you had any time to look at this?

lukaskremla, 2025-04-22:
0.5.0 brings serial connection monitoring. As soon as you disconnect the device the plugin should detect it and disconnect from the device.

closed 2025-04-22 by lukaskremla