lukaskremla/micropython-tools-jetbrains #15: while flashing the ide shows no device connected

nnnayeem, 2025-03-07:
I have rp2040 board. When I try to flash by serical communication after some time error msg shows timeout. Then in the file tree shows  no board is connected. If I try to connect again filesystem does not update even though there is no error msg.
I can verify the device is connected even though msg is device is not connected. If I try to connect with thonny it works fine withoug replugging the device.

nnnayeem, 2025-03-14:
Sorry for the late reply. This is the video link:
https://streamable.com/ulhnqm.

The behaviour this time is actually different. Previously it was not even uploading.

lukaskremla, 2025-03-18:
Hello, thank you for the reports! I've been replicate the issue, but it doesn't seem like something that happens consistently.

It's likely a problem in the terminal widget library that the plugin uses. The 0.3.0 split one large script into several simpler script executions to reduce the performance load on the connected microcontroller.

I've added some delays to the script executions, it seems like that might've fixed it.

This is a pre-release of the 0.4.0, it's not yet fully ready to be released to the marketplace, but it should work. Some things changed and the documentation is not yet written, you will need to use the cyan MicroPython Sources Roots instead of normal Sources Roots.

You'll need to install the plugin manually from disk in the plugins marketplace menu of PyCharm. If you run into any problems or have any questions, feel free to ask, hope it help!

lukaskremla, 2025-03-29:
This issue should be fixed in 0.4.0 and later versions. If you still experience it feel free to open a new issue.

closed 2025-03-29 by lukaskremla

lukaskremla, 2025-06-14:
> Sorry for the late reply. After update file upload works most of the time works fine. but sometimes some folder or file upload takes time and fails. for example I was trying to upload .ttf file and was taking time then giving me Upload I/O error - Device aborted raw paste mode. I tried uploading full forder and individual ttf file. but unfortunately I falied to upload.

I've noticed this sometimes happening as well - generally, when it does happen, the device went through multiple uploads already, disconnecting it and reconnecting it makes the problem go away.

If you find a concrete way to reproduce the bug, please open a new issue and I will fix it. I plan on revisiting and re-testing all parts of the plugin this month and I'll try to see if there's a way to mitigate this issue