lukaskremla/micropython-tools-jetbrains #17: Add MicroPython firmware flashing functionality

lukaskremla, 2025-03-15:
### Discussed in https://github.com/lukaskremla/micropython-tools-jetbrains/discussions/10

<div type='discussions-op-text'>

<sup>Originally posted by **Witawat** March  5, 2025</sup>
Is there any chance to have feature to flash fw in this plugin?

Now I use Thony to flash fw first, then develop program through this plugin.

I hope there will be tool to support flash fw.</div>

Josverl, 2025-03-18:
@lukaskremla , 
I just stumbled on this, and was just thinking on formalizing an API for mpflash. 
currently its only in the form a basic test so that I know ahead when I break things.

Let me know what parts you would like to access, and then lets keep these stable

https://github.com/Josverl/mpflash/blob/main/tests/test_mpf_api.py

lukaskremla, 2025-03-19:
> [@lukaskremla](https://github.com/lukaskremla) , I just stumbled on this, and was just thinking on formalizing an API for mpflash. currently its only in the form a basic test so that I know ahead [...]

Given that the plugin is written in Kotlin integrating with Python code is a bit complicated. 

I'll have to run the Python scripts as a separate process. For mpflash I'm not yet sure how I want to approach this, I'll need some time to get familiar with mpflash and to think through what the implementation should look like. 

There are some things in the plugin that I have to get sorted first. I plan on adding a third "Device" widget to the plugin's tool, which is where I'd like to place the button for opening the firmware flashing dialog. This widget extension requires a modification of the code structure.

But keeping the parts of mpflash I'll be relying on stable is certainly going to be nice! I'll come back to this once I'm done with the prerequisites

Josverl, 2025-03-19:
Assuming you can run a commandline tool then it's a question of what primitive commands and queries you need.
`Mpflash list --json` should be easy to process, it should be possible to add similar to other commands as well