lukaskremla/micropython-tools-jetbrains #18: Make an "Execute in REPL" run configuration

lukaskremla, 2025-03-15:
Hi Lukas,
This is the stubs file I'm using for my pico2 is micropython-v1_24_1-rp2-rpi_pico2-stubs, and for my bog standard pico, micropython-v1_24_1-rp2-rpi_pico-stubs. They both do the same thing.
And thanks for the pointer to the "run this file" option. Exactly what I was looking for!! Now if there was just some way to add it to the run function in the top bar...
;-) Ian

_Originally posted by @mogplus88 in https://github.com/lukaskremla/micropython-tools-jetbrains/discussions/6#discussioncomment-12270484_

lukaskremla, 2025-03-15:
Implemented, will be released in 0.4.0

closed 2025-03-15 by lukaskremla

mogplus88, 2025-03-16:
you're a legend!

;-)

lukaskremla, 2025-03-18:
> **Note** : There are more updates coming to the stubs, as with the additional details in parameters there are also more things to tidy up... if you package them it makes sense to set an alert on the releases. and update the packaging to use ~=1.24.1 rather than ==1.24.1 as the full version is currently1.24.1.post2 https://github.com/Josverl/micropython-stubs/releases
> 
> I recommend **against** copying the packages directly from the repo. things will most certainly break in the near future as I move to different packaging tools.

I agree that copying the packages from your repo wasn't a good way to go. I plan on using [this script](https://github.com/lukaskremla/micropython-tools-jetbrains/blob/main/project-scripts/update_stubs.py) for new versions of this plugin (0.4.0+). 

I'm "watching" your stubs repository's releases and I'll be updating the bundled stubs when a new release goes up.

Josverl, 2025-03-18:
Moving to : https://github.com/lukaskremla/micropython-tools-jetbrains/discussions/20