lukaskremla/micropython-tools-jetbrains #19: Error While uploading files to ESP8266

drakescherer, 2025-03-17:
**Describe the bug**
A clear and concise description of what the bug is.
Esp8266 d1 mini variant has been flashed with micropython successfully, however when trying to use pycharm to upload, results in an error.

**To Reproduce**
Steps to reproduce the behavior:
1. Have an ESP8266 connected via usb
2. confirm repl works in the IDE
3. attempt to upload and run hello world code
4. presented with error about crc32 not being a valid attribute of module

**Expected behavior**
Upload the code and run as normal

**Screenshots or a Video**
Providing a video of the bug depicting how to reproduce it will allow me to fix it significantly faster. Do not have.

**Desktop (please provide the following information):**
 - OS: Win11 Pro
 - IDE: PyCharm Professional
 - IDE Version: 2025.2

**MicroPython Board information (please complete the following information):**
 - Board: ESP8266 D1 Mini Variant
 - Version: 1.24.0

**MicroPython Tools plugin**
 - Version: 0.3.0
 - Used connection type: Serial

**Additional context**
Add any other context about the problem here.

lukaskremla, 2025-03-18:
This is a pre-release of the 0.4.0, it's not yet fully ready to be released to the marketplace, but it should work. Some things changed and the documentation for it is not yet written, you will need to use the cyan MicroPython Sources Roots instead of normal Sources Roots.

You'll need to install the plugin manually from disk in the plugins marketplace menu of PyCharm. If you run into any problems or have any questions, feel free to ask, hope it helps!

Having issues uploading the file... Github refuses to accept the upload, https://drive.proton.me/urls/7B1TGMZJ0W#l3c2mPQujTSt

drakescherer, 2025-03-18:
I attempted to do this, but had an issue: No folders were marked as MicroPython Sources Roots

I then attempted to mark the project folder as a MicroPython Source Root and it gave me another error: 
`Upload error - RuntimeExceptionWithAttachments: Access is allowed from Event Dispatch Thread (EDT) only; If you access or modify model on EDT consider wrapping your code in WriteIntentReadAction or ReadAction; see https://jb.gg/ij-platform-threading for details Current thread: Thread[#695,DefaultDispatcher-worker-45,5,main] 1516328269 (EventQueue.isDispatchThread()=false) SystemEventQueueThread: Thread[#64,AWT-EventQueue-0,6,main] 2100616536`

I do not know where else to mark as root, as the option does not show up in the external libraries folder for micropython.

drakescherer, 2025-03-18:
I did that and it works just fine now, did not have to do that with my other project using esp32 though, so not sure why I had to do it now.

lukaskremla, 2025-03-19:
> I did that and it works just fine now, did not have to do that with my other project using esp32 though, so not sure why I had to do it now.

If you created the other project before installing the 0.4.0 version then that's understandable, since MicroPython Sources Roots were first added in 0.4.0 and the upload run configurations were reworked.

The reason for this is that a custom folder marking allows the plugin to work with all Jetbrains IDEs that support python, it simplifies the upload algorithm, allows better project organization and makes it clearer how things get uploaded, since you can think of the top most MicroPython sources folders as the root "/" of your device's file system. 

I'll write new documentation and add some comments to the run configuration that will explain the new behavior. Hopefully I'll be able to release 0.4.1 this week, your IDE will let you know when the update is out.

If you have any other questions, issues or suggestions, feel free to open another issue or start a discussion!

closed 2025-03-19 by lukaskremla