lukaskremla/micropython-tools-jetbrains #21: latest plugin fails to upload/flash

rlalance, 2025-03-26:
**Describe the bug**
The last version of the pycharm plugin is failing right after flashing main.py.

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Create a run configuration to flash main.py 
[image]
2. Run the configuration: 

[image]

5. Upload seems to work but end up with a bunch of errors.

**Screenshots or a Video**
[image]

**Desktop (please provide the following information):**
Windows 11.0
PyCharm 2024.3.4 (Professional Edition)
Build #PY-243.25659.43, built on February 27, 2025
Runtime version: 21.0.6+8-b631.39 amd64 (JCEF 122.1.9)
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
Toolkit: sun.awt.windows.WToolkit
GC: G1 Young Generation, G1 Concurrent GC, G1 Old Generation
Memory: 2048M
Cores: 14

**MicroPython Board information (please complete the following information):**
 - Board: RP Pico
 - Version: Raspberry Pi Pico with RP2040
 - MicroPython v1.24.1 on 2024-11-29; 

**MicroPython Tools plugin**
 - Version:

[image]

 - Used connection type USB-COM8

**Additional context**
This worked before I updated pycharm and the plugin. So the previous version of pycharm and micropython worked great!

rlalance, 2025-03-26:
I'm going to try to go back to my previous version of pycharm to see if the plugin works there.

lukaskremla, 2025-03-26:
Hello,

I've tried the latest 0.4.1 version available from the plugin marketplace with a raspberry pi pico W running MicroPython 1.24.1 and I haven't been able to replicate your issue.

Could you please show me the current project structure you have setup? The name of the configuration shouldn't have the entire path in it (as long as it was auto-generated by the plugin). You could also try creating a fresh project and seeing if that helps. 

Additionally, I would recommend against using the "path" option unless it's somehow necessary to achieve a workflow you have setup with other run configurations. I'd recommend you to create a folder in your project (name doesn't matter), mark it as a "MicroPython Sources Root" through the right-click menu and then select the "Project" option in the run configuration. You can then treat the top most MicroPython Sources Roots as the root "/" of your MicroPython device when structuring the project. There's more on this in this repo's readme.

Please try this and let me know how it goes.

rlalance, 2025-03-26:
Works with earlier version, even if it is copied to the wrong folder...

[image]

[image]

lukaskremla, 2025-03-29:
The screenshots you have sent are of a different plugin developed by JetBrains, which is no longer maintained or updated.

Even for your use case I would recommend marking the directory which contains the MicroPython files as a "MicroPython Sources Root". That is what the "mark as" functionality is designed for, to allow combined projects with both MicroPython and non-MicroPython code.

That being said, the decision of which run configuration type to use is up to you, both should work. 

I have managed to replicate your issue with more thorough testing. The run configuration would attempt to upload the file with a path relative to the project directory without creating the containing directories, so it always failed with an error. 

I have temporarily modified the run configuration to always upload to the root of the device, in one of the future versions I will add another option to specify not only which file to upload but what target path it should be uploaded with.

As of writing this reply the 0.4.2 version which implements this fix is pending approval to be publicly available in the JetBrains plugin marketplace. Until it is approved you can download the archive and manually install it from the [0.4.2 release](https://github.com/lukaskremla/micropython-tools-jetbrains/releases/tag/0.4.2).

More info on how to do this can be found [here](https://www.jetbrains.com/help/mps/managing-plugins.html#install_plugin_from_disk) the option is called "Install Plugin from disk".

rlalance, 2025-03-29:
This is awesome Lukas! Thank you!

lukaskremla, 2025-04-22:
The issue was solved in 0.4.2

closed 2025-04-22 by lukaskremla