lukaskremla/micropython-tools-jetbrains #23: 2025.1 support

Trilife, 2025-04-16:
Love this plug-in! Thanks so much.

Pycharm 2025.1 just dropped. Is the plug-in compatible? If not, I'll wait to upgrade to 2025.1 until the plug-in is ready.

Cheers!

lukaskremla, 2025-04-16:
Hello,

Thanks for the kind words! Happy to have developed a tool you find useful!

The latest public 0.4.2 version is not available in 2025.1 IDEs. A newer 0.4.3 is pending approval by JeBrains, hopefully it will be published soon.

I thought I'd be able to get the 2025.1 compatibility sorted with 0.5.0, however, I underestimated the amount of time its development will take, so I didn't have an update ready in time.

Sorry for the delay. Hope to avoid this in the future when new IDE versions drop.

I’ll follow up here when 0.4.3 goes live.

lukaskremla, 2025-04-17:
@Trilife 

0.4.3 is now publicly available in the JetBrains plugin marketplace of 2025.1 IDEs.

closed 2025-04-17 by lukaskremla