lukaskremla/micropython-tools-jetbrains #25: Latest MicroPython 1.25.0 stubs not found in settings / Bump micropython-stubs

maghiel, 2025-04-20:
**Describe the bug**
Not sure if this is a bug or an enchancement request, but: 
In the plugin settings I am unable to find stubs for the latest MicroPython version 1.25.0. 
Jos Verlinde has updated and released micropython-stubs v1.25.0 (https://github.com/Josverl/micropython-stubs) three days ago. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Settings > Languages & Frameworks > MicroPython Tools'
2. In 'MicroPython Stubs > Stubs package' type '25'

**Expected behavior**
All stubs with version 1.25.0 are found

**Actual result**
No results

**Desktop (please provide the following information):**
 - OS: [Windows]
 - IDE [PyCharm Community]
 - IDE Version [2025.1]

**MicroPython Board information (please complete the following information):**
Unrelated

**MicroPython Tools plugin**
 - Version: [0.4.3]

lukaskremla, 2025-04-22:
The new 0.5.0 release is now publicly available in the plugin marketplace. It contains all currently available MicroPython 1.25.0 stubs

closed 2025-04-22 by lukaskremla

maghiel, 2025-04-22:
Works, thank you!