lukaskremla/micropython-tools-jetbrains #26: Crash when resetting (CTRL-D) an active connection in REPL

rirad, 2025-04-24:
**Describe the bug**
When a (WebREPL) connection is active, clicking the Reset button (or pressing CTRL-D) results in a crash.

**To Reproduce**
Steps to reproduce the behavior:
1. Make a WebREPL connection
2. Go to the REPL view
3. Click on the Reset button (small square with round arrow) or press Ctrl-D
4. See error (copied below)

**Expected behavior**
Should not crash. The board is expected to reset itself.

**Desktop (please provide the following information):**
 - OS: Ubuntu
 - IDE PyCharm Professional
 - IDE Version 2025.1]

**MicroPython Board information (please complete the following information):**
 - Board: ESP32 S3
 - Version: 1.24.1

**MicroPython Tools plugin**
 - Version: 0.5.1
 - Used connection type: WebREPL

**Additional context**
```
net.schmizz.sshj.connection.ConnectionException: WebREPL connection failed: Connection refused
	at dev.micropythontools.communication.MpyWebSocketClient$connect$2.invokeSuspend(MpyWebSocketClient.kt:128)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:104)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:608)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:873)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:763)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:750)
```

lukaskremla, 2025-04-24:
Hello,

Thank you for the report. It seems that the recent fix for handling soft resets in WebREPL wasn't perfect.

As a temporary workaround you could try avoiding soft resets and instead try to manually execute main.py in REPL via the right click context menu. It's not perfect, but it should work for now.

I'll try to get it fixed and release a new minor version soon.

rirad, 2025-04-24:
Thanks. My main.py file imports several other modules and running main.py in REPL results in an ImportError. I guess this will work when my code is just contained in a single file. For now I will just use a hard reset on the board itself.

lukaskremla, 2025-04-27:
https://drive.proton.me/urls/HHGQA5W9BM#ORjrJgR6yBty

Above is a link to the pre-release3 of 0.5.2. It contains several bug fixes and improvements including ones to WebREPL.  It can be manually installed [like so](https://www.jetbrains.com/help/mps/managing-plugins.html#install_plugin_from_disk). The reset issue should be resolved. There is also a new UI for configuring WebREPL in the settings, you will have to set it up again.

Handling soft resets in WebREPL communication is troublesome. Whenever you send a soft reset command to the device either through CTRL D or through the plugin's UI the device reinitializes the network interfaces, meaning the WebREPL connection is dropped. This requires careful handling of reconnecting to the device in the background whenever a restart is detected. 

It should work now, so it should be possible to observe code execution with WebREPL again, but - there are some caveats. After soft-resets you might miss early debug prints from your device, because it takes some time to reconnect, you can mitigate this by adding a few seconds of delay after initializing wi-fi and WebREPL to give the plugin time to reconnect. 

Additionally, if you soft reset the device from anywhere in your code, the plugin won't be able to detect it and the WebREPL connection will drop. These are the general limitations of the WebREPL protocol and there's not much I can do about them unless the MicroPython WebREPL implementation changes.

This might not be the case for all ports, but in most cases it will be like this.

As for the import errors you experienced - that's odd and shouldn't happen. If you execute main.py in REPL it will try to resolve imports just as if it was located in the root of the device's file system. Are you sure the imported modules were present at the import paths configured in your main.py?

rirad, 2025-04-27:
I am testing 0.5.2 right now in this is what I observe:

My ESP32-S3 is connected to a p1 port and receives telegrams (a led blinks when a telegram is received every second). 

Step 1:
When I start PyCharm and connect to the WebREPL using 0.5.2, I see the content of the file system and the REPL window shows:

MicroPython v1.24.1-dirty on 2025-01-12; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
>>> 

At this point the board stops receiving telegrams (led stops blinking). Since the led stops blinking, I assume the REPL view is correct and no info is printed to the REPL. I am not sure what causes the board to freeze but I will connect it with USB later and see if I get some hints there. 

Step 2:
Now if I press the Reset button, I get the message that it is "Communicating with the board". Once that is done I do get output from the WebREPL:
>>> INFO     lib.p1     Suppress send: 58 sec till publication of telegram 1.
INFO     lib.p1     Suppress send: 57 sec till publication of telegram 1.
INFO     lib.p1     Suppress send: 56 sec till publication of telegram 1.
INFO     lib.p1     Suppress send: 55 sec till publication of telegram 1.
INFO     lib.p1     Suppress send: 54 sec till publication of telegram 1.
INFO     lib.p1     Suppress send: 53 sec till publication of telegram 1.
INFO     lib.p1     Suppress send: 52 sec till publication of telegram 1.

At this point, the LED also starts blinking again indicating the board receives telegrams.

Step 3:
Now if I press the Reset button again, I don't get the "Communicating with the board" dialog window (or very briefly) and the REPL window is cleared and only shows a blinking cursor and the board stopped receiving telegrams.

Step 4:
When I press the reset button for a third time, I again get the "Communicating with the board" dialog which then results in an error:
Reset I/O error - WebREPL connection failed: Connection refused
At this point, the board does start to receive telegrams again.

Step 5:
Due to the error, the WebREPL connection is lost and I need to reconnect. The Telegram led is blinking thus the board receives telegrams. I press the plug icon to make a new connection and I get the file system content again and in the REPL view:

MicroPython v1.24.1-dirty on 2025-01-12; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
>>> 

At this point, the board stops receiving telegrams again (like in step 1)

Step 6: 
If I press the reset button in the REPL view again (like in step 2), I immediately get a error:
Reset I/O error - WebREPL connection failed: Connection refused

I do think the board did receive the reset since the telegrams are received again.

Step 7:
Since the board seems to be up and running (pre step 1 condition). I restart PyCharm and connect again to the WebREPL. The results are like step 1.

Step 8:
I press the reset button and now I get:
Reset timed out
Connection attempt timed out

And the connection is closed.

rirad, 2025-04-27:
When I connect to WebREPL interface from micropython, after providing the password, this is what I get:
[image]
Making the connection in the Micropython WebREPL web interface does not seem to reset the board upon making the connection (as a reset would also reset the telegram counter).

This is very similar to what I get after step 2 (but here the telegram counter starts at 1 since a reset was initiated). I assume the problem that remains is that for some reason making the WebREPL connection in PyCharm freezes/stops the board which can be fixed with the resetting the board in the REPL view again. However, subsequent Resets don't work. Once I tested these steps with the USB REPL connected and I get some hints what happens there, I will report.

rirad, 2025-04-27:
> It should work now, so it should be possible to observe code execution with WebREPL again, but - there are some caveats. After soft-resets you might miss early debug prints from your device, because [...]

Yes that makes sense and is not an issue. 

> Additionally, if you soft reset the device from anywhere in your code, the plugin won't be able to detect it and the WebREPL connection will drop. These are the general limitations of the WebREPL [...]

My code is heavily inspired by Josverl project [p1-meter](https://github.com/Josverl/micropython-p1meter). Here the board can be reset in case of unexpected issues. However, this does not happen frequently and it makes sense the connection is lost in those cases. The MicroPython WebREPL has the same issue and is not automatically reconnecting in that case.

>As for the import errors you experienced - that's odd and shouldn't happen. If you execute main.py in REPL it will try to resolve imports just as if it was located in the root of the device's file system. Are you sure the imported modules were present at the import paths configured in your main.py?

Yes I think I used a different project than the project that was flashed on the ESP by mistake (not sure anymore). But thanks for confirming that it should resolve imports.

lukaskremla, 2025-04-27:
The plugin is actually intentionally interrupting the code execution on the device in order to provide its features such as scanning the file system, it needs to run MicroPython scripts on the board to retrieve this information, hence interrupting the code execution.

As with serial REPL, you should be able to get the code executing again by performing a soft reset. From your description it appears that you're only able to successfully perform a soft reset once, which is a bug.

I'll spend some more time debugging WebREPL and see if I can make the resets work more consistently

rirad, 2025-04-30:
Hi Lukas, I just tested version 0.5.3.

My script was starting a WiFi connection and subsequently a WebRepl connection in main.py. I was not checking if the WebRepl was already running. Thus after making a connection and performing a soft reset, my script would start a new WebRepl. By doing so the current WebRepl could be closed resulting in a disconnect error in PyCharm (thus effectively preventing me from seeing the debug information).

I now first check if I can connect to the WebRepl. If I can, I will not start a new WebRepl. If I get an OSERROR, I can't connect to the WebRepl and a new WebREPL will be started.

This seem to work robustly. Now if I start a WebRepl connection and press the reset button (or CTRL-D), the WiFi and WebRepl remain active, I skip starting a new WebREPL and the REPL view shows the correct debug information.

So as far as I can tell, the issue is solved.

lukaskremla, 2025-05-03:
Hello,

Thanks for the follow up! Glad to hear it is working for you now.

WebREPL as a protocol isn't great. It's implemented via WebSockets for browser compatibility (since they can't open raw socket connections). It also doesn't seem to adhere to the WebSocket specification so I had to frankenstein a Java WebSocket library to get things working the way they are now. There are also several bugs in it that have been around for years and it is incredibly slow.

Recently I saw someone implementing custom bluetooth REPL utilizing the os.dupterm(). That could be a universal pure-python way to implement custom remote development that doesn't suffer from the issues that WebREPL does. It would require more setup - downloading a custom library - than WebREPL does, but it should be faster and if done right more reliable.

I might investigate that later after the more important features like MicroPython flashing and mpy-cross integration are implemented. If you run into any other issues using WebREPL though, feel free to open another issue. 

I assume #27 is also resolved now?

lukaskremla, 2025-05-08:
Thank you for reporting the bug and testing the updates! If you run into any other problems feel free to open an issue again.

closed 2025-05-08 by lukaskremla