lukaskremla/micropython-tools-jetbrains #28: Add MicroPython package manager

maghiel, 2025-04-25:
Not sure how doable it is, but wouldn't it be awesome if this plugin had some kind of MicroPython package manager? 
Generally, I am really fond of using your plugin, but the lack of any package manager takes me out of my flow.

Right now we have to resort to using mip through REPL / using PyCharms package management and copying the packages over / manually copying packages, etc.

For me it doesn't need to be a manager that works directly on the device like Thonny does; I'm thinking more of a some kind of workflow that manages packages in the lib subdir of a dir marked as MicroPython Sources Root.

Anyway, keep up the great work!

lukaskremla, 2025-04-26:
Hello,

This is an interesting suggestion. I haven't worked with the mip package manager in any of my projects yet, but it does seem like something that's growing in popularity and is worth supporting.

I find installing packages directly to the device, without going through the IDE project first a bad practice, because you don't have a back up if something happens to the device, you can't easily review the package source code and it bypasses the git integration. So I agree with you that this should be some kind of a project MicroPython package manager.

Perhaps I could add another action to the mark folder as... group "MicroPython Library Root" or something like that, then you'd have some UI for actually downloading the packages and you'd be able to choose the destination i.e. the Library Root.

I'll consider this, I do think it'd be nice to have this in the plugin. Though I can't say when I'll get to this. Currently addressing any bugs that might come up and adding MicroPython firmware flashing, mpy-cross integration and a better stub manager are priorities. This is also an open source project that I work on in my free time - so I can't give you any guarantees about when I might get to this.

maghiel, 2025-04-26:
> Perhaps I could add another action to the mark folder as... group "MicroPython Library Root" or something like that, then you'd have some UI for actually downloading the packages and you'd be able to choose the destination i.e. the Library Root.

Great idea!

> I'll consider this, I do think it'd be nice to have this in the plugin. Though I can't say when I'll get to this. Currently addressing any bugs that might come up and adding MicroPython firmware [...]

Ofcourse and I understand :)

lukaskremla, 2025-11-10:
https://github.com/orgs/micropython/discussions/10255

This discussion talks about using the micropython unix port to install mip packages in your terminal/IDE and not relying on using a connected MicroPython device.

With the current state of mip, it doesn't seem worthwhile to implement a package manager in the plugin, there are higher priority assignments right now - such as #17.

I'm open to reconsidering this decision in the future, as of right now this is not a planned addition to the plugin.

closed 2025-11-10 by lukaskremla