lukaskremla/micropython-tools-jetbrains #30: ESP32 DevKit V4

anyone14959, 2025-04-26:
Hi there,
since v0.5x it's not possible anymore to establish a connection, neither by COM nor by webrepl....

Nedded to downgrade

**To Reproduce**
Steps to reproduce the behavior:
1. Have a ESP32 ready with preinstalled MicroPython
2. Try to connect
3. No connection possible

**Expected behavior**
Connection as with 0.4.x Versions

**Desktop (please provide the following information):**
 - OS: Windows
 - IDE [e.g. PyCharm Community
 - IDE Version 2025.1

**MicroPython Board information (please complete the following information):**
 - Board: [e.g. ESP32 DevKit4
 - Version: independant

**MicroPython Tools plugin**
 - Version: 0.5.x
 - Used connection type: independant

lukaskremla, 2025-04-26:
Hello,

What exactly happens? Could you maybe record the behavior you're seeing? Do you get shown an error message of some kind, does it just freeze?

Also, you have tried the 0.5.1 plugin version with MicroPython versions of at least 1.24.1 or 1.25.0?

anyone14959, 2025-04-26:
[Aufzeichnung 2025-04-26 091314.zip](https://github.com/user-attachments/files/19921100/Aufzeichnung.2025-04-26.091314.zip)
I made a small record, showing what happens. 
1st with 0.4.3 everything is working as expected.
Then afer updating to 0.5.1 its trying to connect withot any respond.

When in REPL view, I see messages from the board (not recorded) but I guess there is some communication ongoing

anyone14959, 2025-04-26:
[Aufzeichnung 2025-04-26 092034.zip](https://github.com/user-attachments/files/19921119/Aufzeichnung.2025-04-26.092034.zip)
Record of the REPL view while trying to connect

lukaskremla, 2025-04-26:
Okay, I have set the latest IDE and plugin version up on my windows machine. I did run into the issue you're describing, but I can't replicate it consistently. After start-up I'm unable to connect, after 1-2 attempts I can connect, and then I can't again for a few attempts rinse and repeat.

You should be able to use the latest version if you keep trying to connect until you succeed. 

I will look into what's causing this and release a patch as soon as I get it fixed.

anyone14959, 2025-04-26:
Belive me I tried it consistently for an hour or so, it's impossible with my system.
But if you want test a beta or so, I'm at your disposal...

lukaskremla, 2025-04-26:
> Belive me I tried it consistently for an hour or so, it's impossible with my system. But if you want test a beta or so, I'm at your disposal...

It seems that the issue might be with connection attempts timing out. There was an issue in the code that prevented such errors from getting reported.

https://drive.proton.me/urls/WEWQBHKG0R#sl4tlSffY3cl

Uploading the archive to GitHub keeps failing, so I had to upload it to the link above. You can [manually install](https://www.jetbrains.com/help/mps/managing-plugins.html#install_plugin_from_disk) this pre-release version.

Let me know if it helps.

anyone14959, 2025-04-26:
Works (mainly) just the stop button (red square) is a litte weird, when pressing is ask me to disconnect from COm, and it does it, but the it's not possible to reconnect - need to restart the full IDE to reconnect

lukaskremla, 2025-04-26:
Thanks for trying it out! This bug is apparently quite peculiar, so I really appreciate you helping me fix it.

Could you please record the odd behavior of the stop button as well as what happens after you disconnect? It would help me pin the issue down and fix it.

anyone14959, 2025-04-27:
[Aufzeichnung 2025-04-27 074516.zip](https://github.com/user-attachments/files/19927672/Aufzeichnung.2025-04-27.074516.zip)

Good morning Lukas, enclosed the recording of the behavior of the stop button. The first seconsd just demonstration that the connect button works, but when pressing the stop-button the issue starts...

anyone14959, 2025-04-27:
[Aufzeichnung 2025-04-27 075907.zip](https://github.com/user-attachments/files/19927735/Aufzeichnung.2025-04-27.075907.zip)
Hi Lukas, some additional information:
instead of restarting the IDE it is sufficient to disconnect and reconnect the ESP32 to the interface.
Further, (see recording) the stop button is indeed doing something, (stopping the running script) but also disconnecting the COM and probably leaving the connection open.

lukaskremla, 2025-04-27:
Great catch! There was some inconsistent state handling that I missed while finalizing the communication layer rewrite of 0.5.0. The plugin would be left in a state meant for executing its own scripts (file system refresh, uploads etc.) which is a state in between disconnected and connected, the UI is supposed to be under a modal progress dialog in this state, but it wasn't. 

I've fixed the state handling of the interrupt action and uploaded a new pre-release https://drive.proton.me/urls/H62SMMCYBC#QU40ZGP1Df9d. Let me know if you run into any other issues.

Thanks again for the bug reports!

anyone14959, 2025-04-27:
Great job, works now perfectly.
Just one little hint, about WebREPL connection, I was using / configuring it month ago just with standard port, but I found there is no obvious hint anywhere about the standard port 8266. Maybe it would be a nice to have, if user isn't specifying a port, to autoaticly use the standard port? or at least, give a redlined hint if no port is specified by user. At the moment you're just giving the hint if user missed to specify ws://

Hmm, mayby the perfect solution would be if just the IP or Hostname is given to automatically surround it by "ws://" + hostname + ":8266"

What do you think?

But anyway, thank you for this great little plugin

anyone14959, 2025-04-27:
Other ovservation I made: when downloading a file (from ESP to System), a dialog is opened where to store it, but I'd rather expected that it'd be downloded directly into the opened pycharm project folder?!

lukaskremla, 2025-04-27:
I like your idea for improving the WebREPL URL configuration experience. WebREPL doesn't work with wss, not to my knowledge at least, so that bit can be hard-coded as a prefix label of "ws://". The port can be changed by passing a different one to the `webrepl.start()` method, so ti must remain configurable, however, I'll consider whether it might be better to split into two textfields, one for the URL and another for the port. Either way there should be some tooltip or better validation text explaining the default WebREPL port.

Regarding the download dialog - there must always be one, because downloading the files from the board is a potentially destructive operation. If you're downloading main.py into your project root and you already have such file there it will get overwritten - this might be intentional, but it should be confirmed by the user. And there are use cases where you might save it elsewhere too. However, I'll modify the download dialog to not open some random default directory but to open the project root directory by default - if this is what you meant by your suggestion then I apologize for overexplaining.

lukaskremla, 2025-04-27:
https://drive.proton.me/urls/HHGQA5W9BM#ORjrJgR6yBty

Here is another pre-release with the aforementioned changes

anyone14959, 2025-04-27:
So even better, your idea to set the defaut filesave location to the project dir, because both is possibe, easy downloading to the current project if I want to and also the possibility the save it somwhere else.

So looking forward for these nice improvements...

anyone14959, 2025-04-27:
great, just tested it, works fine and pretty comprehensice in the settings!

closed 2025-04-27 by lukaskremla