lukaskremla/micropython-tools-jetbrains #34: Add mpy-cross compilation support

lukaskremla, 2025-05-08:


edel-macias-cubix, 2025-05-12:
This would be a really nice to have!

lukaskremla, 2025-05-12:
> This would be a really nice to have!

I’m planning to release a documentation and bugfix update as version 0.5.4 first. After that, version 0.6.0 will include an mpy-cross run configuration with automatic parameter detection for the connected board. And thanks for the sponsorship - it means a lot!

lukaskremla, 2025-10-16:
This feature is now complete and will become available in the 0.7.0 release on 17.10.2025.

It will be a paid feature - more information can be found here #76 

Apologies for it taking so long to complete this feature - I've gotten side-tracked on other fixes, improvements and projects.

@edel-macias-cubix As you're an early supporter, you're eligible for free access to the pro features - more info in the mentioned discussion #76

closed 2025-10-16 by lukaskremla