lukaskremla/micropython-tools-jetbrains #36: from machine import Pin not find

itmeicn, 2025-05-11:
[image]

**Desktop (please provide the following information):**
 - OS: Windows 11
 - IDE  PyCharm Professional
 - IDE Version 2024.3.5

**MicroPython Board information (please complete the following information):**
 - Board: [ ESP32, 
 - Version: [e.g. 1.24.1]

**MicroPython Tools plugin**
 - Version: 0.5.3
 - Used connection typeSerial

**Additional context**
I used MicroPython tools，bug  machine not find,how to ?

lukaskremla, 2025-05-11:
Have you configured a stub package?

[image]

itmeicn, 2025-05-12:
All right. But there is no syntax prompt when entering characters

bc-109, 2025-05-12:
Hi,

I also have lots of highlighting errors. I already had them with the official plugin. That doesn't bother me much, because the code is working :-) And I never spent much time trying to find what's going wrong...

A stub is defined (micropython-rp2-stubs_1.24.1)

### Examples on standard libraries :

[image]

[image]

### Example on personal modules. 
The root dir (/target) is marked as Micropython  sources root, but subfolders do not seem to be parsed to scan for modules and functions. 

For example, here's an excerpt of boot.py : 

[image]

Here's the tree structure : 

[image]

**lprint()** is defined in **mpinit.py**, which is in /target/src

lukaskremla, 2025-05-12:
> All right. But there is no syntax prompt when entering characters

Do you mean no syntax prompt in the stub package field? Type "micropython" and you should be able to see all auto-completable stub packages.

lukaskremla, 2025-05-12:
@bc-109 

It almost seems like the stubs actually aren't activated. If the stub package is changed right after the IDE starts up, while indexing is running, the settings can say that it was activated, but it actually wasn't. The same thing happens if the `.idea` library files are modified manually.

You could try to switch to a different stub package, apply it, close the settings, wait a bit and then try to activate the desired stub package again.

Both the u-module imports and `from machine import RTC` are valid syntax and the `micropython-rp2-stubs_1.24.1` covers them.

This behavior is a limitation of the current stub package management, in the future I'd like to have a full list/table of available stub packages, allowing users to download them and update them straight from the `pip` source and also make the "current stub package" reflect the actual stub package currently selected in the project.

And for the imports of your own modules - this is out of the scope of this plugin, it's standard PyCharm behavior. I'm not sure if I can influence it from the plugin or via the MicroPython Sources Roots. PyCharm should recognize it if you do `from src import mpinit` or some similar syntax that specifies the `src` sub folder as well

itmeicn, 2025-05-14:
[image]
I inputed Pin.  not  OUT/IN.Many modules are the same

lukaskremla, 2025-05-15:
> [image] I inputed Pin. not OUT/IN.Many modules are the same

What stub package is causing that issue for you?

itmeicn, 2025-05-15:
Ths,i used micropython-stm32-stubs_1.25.0 and python3.11  is ok.

closed 2025-05-15 by itmeicn