lukaskremla/micropython-tools-jetbrains #38: MicroPython `stdlib` stubs are installed but not used

nickovs, 2025-05-19:
**Describe the bug**

When enabling the MicroPython support on a project, PyCharm correctly uses the type stubs installed from the build-specific stubs package, but it fails to use the stubs from the generic MicroPython `stdlib` package. As a result PyCharm correctly handles the interface checking for MicroPython packages but can not find the MicroPython-specific extensions to the standard libraries, and it incorrectly fails to flag an error on inclusion of CPython standard library features that are absent from MicroPython.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new pure Python project in PyCharm
2. Create a new environment
3. In Settings->Languages & Frameworks->MicroPython Tools, click Enable MicroPython support and select the stubs package: `micropython-rp2-rpi_pico2_w-stubs_1.25.0`
4. Create the file `test.py` and insert the following code:
```python
# Headers specific to MicroPython show up correctly
from machine import Pin
from rp2 import PIO, DMA

# Imports of functions that exist in MicroPython versions of
# stdlib but not in CPython are flaged as errors
from asyncio import sleep_ms

# Imports from stdlib modules of entries that don't exist in MicroPython pass
from asyncio import MultiLoopChildWatcher
from json import JSONDecoder
from io import IncrementalNewlineDecoder
```

5. While the imports from `machine` and `rp2` are handled correctly, the import of `sleep_ms` from `asyncio` is flagged as symbol that can't be found and none of the other imports are flagged even though these symbols are missing from MicroPython and from the MicroPython `stdlib` stub package. See the image below for details.

**Expected behavior**
The plugin should make both the port-specific stubs and the `stdlib` stubs available to PyCharm such that PyCharm uses both sets, and thus correctly checks the interfaces and typing for the MicroPython APIs.

**Screenshots or a Video**

A demonstration of the incorrect checking:

[image]

**Desktop (please provide the following information):**
 - OS: macOS
 - IDE: PyCharm Professional
 - IDE Version: 2025.1.1.1

**MicroPython Board information (please complete the following information):**
 - Board: Pico 2W
 - Version: 1.25.0

**MicroPython Tools plugin**
 - Version: 0.5.3
 - Used connection type: Serial/USB

**Additional context**

In the image above you will see that replacing `from asyncio import sleep_ms` with `from stdlib.asyncio import sleep_ms` allows PyCharm to find the correct stub file. This suggests that the problem is related to the setting of the correct search path for the stubs.

Josverl, 2025-05-19:
I could not resist taking a stab but could not find a way to mark the stdlib folder as [Sources Root]

so it requires a bit of an ugly hack :
- ` uv pip install micropython-stdlib-stubs --target my_typeshed`
- then mark `my_typeshed/stdlib` as [Sources Root] 

[image]

This works to override asyncio , but is not practical to repeat for each project.

lukaskremla, 2025-05-20:
Thanks for reporting this, I didn't notice the `stdlib` conflicts up until now.

What @Josverl suggested is seems to be the only way to get custom `stdlib` stubs recognized - for you as a user. I tried installing them directly via `pip install stub_package`, but this doesn't make `stdlib` recognized either.

I'll do some experiments with separating the `stldib` parts of the stubs and attaching both the MicroPython-specific `.pyi` files and the `stdlib` ones individually - this should replicate the working Sources Root approach but programmatically.

lukaskremla, 2025-05-20:
Please try the latest 0.5.4 release. It's publicly available in the plugin marketplace.

lukaskremla, 2025-05-20:
That's not a bad idea - I can quite easily ensure old library selections are invalidated, will do so in the next release. Thanks for the report!

closed 2025-05-20 by lukaskremla