lukaskremla/micropython-tools-jetbrains #40: Error when uploading project

Sabdot33, 2025-05-20:
**Describe the bug**
FormatFlagsConversionMismatchException or Warning when trying to upload my Project:

```
Mode                 LastWriteTime         Length Name                                                                                                                                                                             
----                 -------------         ------ ----
d-----        20/05/2025     20:02                .idea
-a----        20/05/2025     19:29           8311 bme280.py                                                                                                                                                                        
-a----        20/05/2025     19:41            571 main.py
```

**To Reproduce**
Steps to reproduce the behavior:
Not sure but:
Try uploading for the first time:
Get some warning but the uploading works? as far as i know?
Second time uploading:
EITHER
`FormatFlagsConversionMismatchException`
OR 
You get that non-fatal warning again and then the whole IDE freezes...

and from then on always the same.

**Expected behavior**
just do the thing.

**Screenshots or a Video**
Sry for making it a zip
[Recording 2025-05-20 204021.zip](https://github.com/user-attachments/files/20353640/Recording.2025-05-20.204021.zip)

**Desktop (please provide the following information):**
 - OS: Windows
 - IDE: PyCharm Community 
 - IDE Version 2025.1.1.1

**MicroPython Board information (please complete the following information):**
 - Board: Rasp Pi Pico RP2040 V1589

**MicroPython Tools plugin**
 - Version: latest 0.5.4
 - Used connection type: Serial REPL

**Additional context**
Wiring:
Yellow: SCL Green: SDA
Actually i used a BME280 sensor instead of DHT but the wiring is the same.
i can also share my code if you need.
[image]

lukaskremla, 2025-05-20:
Confirmed the bug, working on a fix. I've pulled the version from the marketplace for now, you can downgrade to the stable 0.5.3 release [here](https://github.com/lukaskremla/micropython-tools-jetbrains/releases/tag/0.5.3)

lukaskremla, 2025-05-20:
I've released [0.5.5](https://github.com/lukaskremla/micropython-tools-jetbrains/releases/tag/0.5.5) the bug is fixed in that version. You can download the archive manually from here and install it in the plugin marketplace gear icon menu in your IDE. This new version should become available through the marketplace sometime on the 21.5.

Sabdot33, 2025-05-21:
Thanks for the quick fix!