lukaskremla/micropython-tools-jetbrains #41: MicroPython Sources Root

joaonc, 2025-05-25:
Hello and thank you for this plugin.  Has made life much easier working with MicroPython in PyCharm 🙌 

[This screenshot](https://github.com/lukaskremla/micropython-tools-jetbrains/blob/main/media/micropython_sources.png) notes that there's a way to mark a folder as MicroPython Sources Root, but I don't see it in PyCharm..

Using PyCharm `2025.1.1.1` on Windows 11.

Will setting this root set the Project Root in this second option 👇?

[image]

My project has other folders with code that I want to keep for local development but not copy to the board, which I imagine is a setup many other people have.

[image]

lukaskremla, 2025-05-25:
Do you have the latest 0.5.5 version of the plugin installed and have you enabled MicroPython support for this specific project? 

If so, you should be able to mark the folder as a `MicroPython Sources Root` through the same `Mark as` action you used to mark it as the standard `Sources Root`.

If you verified these two requirements and it still doesn't work - could you try to create a fresh new project, enable MicroPython support for it and try if you can mark the roots there? - This would help rule out whether the issue is specific to the state of your current project's configuration.

If you succeed in marking the the folder as a `MicroPython Sources Root` then the action will switch to only displaying `Upload to device root "/"` in which case the contents of this folder will get uploaded straight to the device file system root.

I also recommend checking run configurations out if you haven't already, those make things easier and offer you some more useful features.

joaonc, 2025-05-25:
Ah!   Thanks for the quick reply.

Indeed in the _Mark Directory as_ it does show, as expected and as the screenshot in the instructions point out 😊.
For some reason, I got stuck in the _Project Structure_ UI (screenshot below), which is what I normally use to mark directories.

[image]

[image]

Thanks again!