lukaskremla/micropython-tools-jetbrains #46: ESP8266 error when connecting

d-marquina, 2025-06-24:
Hi, first of all, thanks for this plugin, I have tested it with a Raspberry Pi Pico and an ESP32-S3 with no problems.
Sadly, when I try to use an old ESP8266 and click the Connect button, I get one error message:

Connect error - RuntimeExceptionWithAttachments: Access is allowed from Event Dispatch Thread (EDT) only; If you access or modify model on EDT consider wrapping your code in WriteIntentReadAction or ReadAction; see https://jb.gg/ij-platform-threading for details Current thread: Thread[#319,DefaultDispatcher-worker-23,6,main] 146110132 (EventQueue.isDispatchThread()=false) SystemEventQueueThread: Thread[#36,AWT-EventQueue-0,10,main] 296626010

After that, I have to click the refresh button in order to load the files inside the ESP8266. Then, I can run and upload scripts like always.

Steps to reproduce the behavior:
1. Flash an ESP8266 with MicroPython 1.25.0
2. Enable MicroPython Tools
3. Click on Connect
4. Error message appears

**Expected behavior**
Load files automatically after clicking the connect button.

**Screenshots or a Video**

[image]

**Desktop (please provide the following information):**
 - OS: Windows
 - IDE: PyCharm Community
 - IDE Version: 2025.1.2

**MicroPython Board information (please complete the following information):**
 - Board: ESP8266
 - Version: 1.25.0

**MicroPython Tools plugin**
 - Version: 0.5.8
 - Used connection type: Serial

lukaskremla, 2025-06-28:
This only happens with the ESP8266 device? If you try the exact same steps with either of the other devices that worked, this doesn't happen?

Have you been able to consistently replicate the issue?

d-marquina, 2025-07-01:
I have tried now with 4 devices: Raspberry Pi Pico, ESP32, ESP32-S3 and ESP8266, the original problem only happens with the last one.
Also, I installed PyCharm on a different PC and only the ESP8266 has that problem.
Additionaly, I must say my ESP8266 needs a CH340 driver, it coul be related.

d-marquina, 2025-07-05:
Thank you, it works as expected now: the warning message appears and the files are correctly listed.
This is the warning dialog:

[image]

lukaskremla, 2025-07-06:
That dialog is something you can expect on most ESP8266 boards, and it displayed correctly.

Thanks for reporting the bug! If you run into any other problems in the future, feel free to open another issue.

closed 2025-07-06 by lukaskremla