lukaskremla/micropython-tools-jetbrains #48: [feature] command history

okram, 2025-06-25:
Can you provide standard REPL history support between uploads? e.g. leveraging packages like `readline` or `rlwrap`.

lukaskremla, 2025-06-28:
Is there something about the way REPL currently handles history with the "auto clear REPL" checkbox disabled?

The REPL widget's behavior isn't perfect, I'm aware of some bugs and inaccuracies in it.

Is there something specific you're missing right now? I'm open to suggestions

okram, 2025-06-29:
Ah. The REPL's "up arrow" works as expected **intra-session**. However, when I upload and reboot, my previous history is no longer available. That is to say, **inter-session** history doesn't exist. 

There is a fine line between what `micropython-tools-jetbrains` can/should do and what the Micropython REPL itself can/should do. I'm not versed in the Python REPL scene (Python being new to me) so I don't know if there is a `.history` file or something of the sort that could be propagated between uploads.