lukaskremla/micropython-tools-jetbrains #50: [ASK]Usage of "Exclude Selected Paths" Feature

qq2661384348, 2025-06-26:
When using the "Exclude selected paths" feature, I entered /test in the Excluded Paths field. There's a test directory under the root directory of my project's source folder, but this directory is still being uploaded to the device.

Am I using it incorrectly? If so, what's the proper way to configure this?

[image]

[image]

lukaskremla, 2025-06-28:
The Exclude selected paths option is related to the "synchronize option". If you specify an excluded path then this specified path won't be deleted on the device, if you run an upload that doesn't modify it.

So if you have synchronization enabled and have a `/logs` directory on the device, the path of which you exclude, and you only upload `main.py`, `boot.py` and a `/lib` directory, then the `/logs` folder won't be deleted (synchronized), which would happen otherwise.

This is useful for keeping persistent config files and logs, that your code generates on the board and that you don't want deleted each time you upload modified source code.

To achieve your goal, you need to restructure your project. Create a folder and name it, for example, `src`, right click it, go to "mark as" and select "MicroPython Sources Root". Now, this `src` folder is going to be considered as the `/` root of your MicroPython device by the upload project run configuration.

So put all of the code and directories you want to upload to the root `/` of your device inside the newly created folder and keep the `test` directory outside of it. The documentation goes more in detail on it [here](https://github.com/lukaskremla/micropython-tools-jetbrains/blob/main/DOCUMENTATION.md#project)

qq2661384348, 2025-06-28:
Ah, that makes sense now. Thanks for the heads-up! I see now that it's for excluding files on the device, not my PC. My mistake!
By the way, I love this plugin you've developed! I was wondering if you might plan to add a feature for compiling to .mpy files? I actually built my own little tool to do that with incremental builds because I needed it so much, but it would be awesome to have it integrated directly into your plugin.
Seriously, thank you for your work. I'm a huge fan of MicroPython, and your plugin has been a massive help in making development so much smoother!

qq2661384348, 2025-06-28:
From my perspective, the current user base of MicroPython isn't particularly large, and the number of people willing to pay for its use would likely be even smaller. However, if a plugin offered features like breakpoint debugging, register inspection, and seamless PyCharm integration comparable to standard Python workflows, I would consider it extremely valuable. Personally, I'd be happy to pay for such functionality. Regarding pricing, even if this feature was worth $10 USD, as an average Chinese college student, I'd actually be reluctant to spend that amount. Here are the key reasons:

1. MicroPython won't become my primary development platform - it's only my choice for rapid prototyping when developing specific embedded devices
2. When it's not absolutely necessary, I'd rather use my custom-built log module for serial output debugging or file logging than pay $10

In summary, while breakpoint debugging is highly appealing to me, I would still prioritize free solutions unless the pricing becomes reasonable enough. After all, if I truly needed these features, I could develop them myself.

lukaskremla, 2025-06-28:
Fully understand that. I too feel 10$ is too high for a MicroPython plugin.

Most likely the monthly price will be 5-7$ a month and when bought as annual something around 4$ per month. This will depend on how many of the planned features I'll be able to add.

Thanks for the feedback!