lukaskremla/micropython-tools-jetbrains #51: Bug Report: Chinese Character Truncation in REPL Output

udreams, 2025-07-18:
**Describe the bug**
In the MicroPython REPL, when printing Chinese text that exceeds the terminal line width, some characters are truncated. This issue does not occur with English text.

**To Reproduce**
Run the following command in the REPL:
print('你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！')

**Expected behavior**
The full Chinese string should be printed without truncation:
你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！你好，世界！
But, in actual, some Chinese characters are missing when the text wraps to a new line.

**Screenshots or a Video**

[image]

**Desktop (please provide the following information):**
 - OS: Windows
 - IDE PyCharm Community
 - IDE Version 2025.1.3

**MicroPython Board information (please complete the following information):**
 - Board: ESP32
 - Version: 1.25.0

**MicroPython Tools plugin**
 - Version: 0.5.10
 - Used connection type Serial

**Additional context**
The issue may be caused by Chinese characters being treated as single-width characters.

udreams, 2025-07-26:
Thank you for looking into this! I really appreciate your time and effort, even if it turns out to be a limitation of the JetBrains terminal widget. Thanks again!

lukaskremla, 2025-07-28:
I can confirm that the problem stems from the `JBTerminalWidget` element that the plugin uses.

CJK characters seem to not be supported by this widget, at least not by default or in a way that would be easy to enable.

I'll try to find out if there is a way to achieve CJK compatibility - I'll have to try to get in touch with JetBrains to find out, I'm not sure how long it may take for them to respond.

udreams, 2025-07-28:
I've also been investigating this issue and wanted to share my findings. After reviewing the [JBTerminalWidget documentation](https://dploeger.github.io/intellij-api-doc/com/intellij/terminal/JBTerminalWidget.html), I noticed there's no explicit mention of CJK character width calculation support in the component.

I tried two approaches to work around this:

1. First, I looked at the alternative implementation used in JetBrains' official MicroPython plugin ([source](https://github.com/JetBrains/intellij-micropython/blob/master/src/main/kotlin/com/jetbrains/micropython/repl/ToolWindowReplTab.kt)) which uses `ShellTerminalWidget` instead. However, after switching to `ShellTerminalWidget`, the issue persisted.

2. Then I attempted enabling AutoWrap mode programmatically with:

```java
setModeEnabled(TerminalMode.AutoWrap, true);
```

But unfortunately this didn't resolve the character truncation either.

These findings suggest the issue might be deeper in the terminal rendering stack. I found a relevant ticket on YouTrack ([IJPL-104924](https://youtrack.jetbrains.com/issue/IJPL-104924)) where other users have reported identical problems with CJK character truncation in the built-in terminal. This suggests it's a known platform-level issue affecting multiple JetBrains products.

The character truncation significantly impacts debugging workflows:

- Forces use of external terminals (like Xshell) for CJK text as a workaroud
- Requires English-only logging as workaround
- Still fails when receiving Chinese debug messages
- Critical information gets truncated during crucial debugging sessions

While we can mitigate by using English outputs, the core issue remains unresolved when handling CJK system responses - a fundamental limitation that needs JetBrains' attention.

[image]

udreams, 2025-08-06:
Based on your suggestion, I've fixed the issue and submitted a PR (#63)—line wrapping now works correctly. However, the REPL shortcuts (Ctrl+A/B/C/D/E) are not functioning. Could you help resolve this? Thanks!

lukaskremla, 2025-08-08:
Thank you for the initiative.

Unfortunately switching to `ConsoleViewImpl` would break more things than it would fix. I've gone into some more detail in the PR.

I will get in touch with JetBrains to see what can be done to add support for CJK.

lukaskremla, 2025-11-05:
@udreams Hello, apologies for the delay. There was a response from JetBrains I missed [here](https://platform.jetbrains.com/t/jbterminalwidget-and-cjk-character-support/2490/2).

The new terminal should be a fix regardless once it is released and I can implement it, however, there are some suggestions of what to try - you can check the steps and see if the recommendation helps?

udreams, 2025-11-05:
Thank you for the helpful suggestions.
Based on my testing, the main issue doesn't seem to be ConPTY truncation, as switching to the ConsoleViewImpl works fine.
I can confirm that the following implementation correctly renders new lines and ANSI colors using ConsoleViewImpl.
[image]
However, I am facing a separate issue with terminal interactivity as you mentioned above. Specifically, arrow keys are not being sent correctly to the MicroPython REPL, which prevents browsing command history.
Maybe, the resolution is now pending the release of the new terminal implementation in an upcoming JetBrains IDE version.

udreams, 2026-02-06:
> I've filed a YouTrack issue for the CJK character wrapping/unwrapping terminal issue.
> 
> You can track it here: https://youtrack.jetbrains.com/issue/IJPL-233083/Classic-terminal-corrupts-CJK-characters-when-wrapping-unwrapping-due-to-resize

Thank you for the update and for filing the YouTrack issue. A current workaround is to use a local terminal emulator (like Xshell) to run commands manually, when dealing with Chinese text.