lukaskremla/micropython-tools-jetbrains #55: Feature request: Buttons to paste text into the console at the right of the plugin window.

ThomasAtBBTF, 2025-07-29:
Hi,

my usecase is, to write "Python code" ;) and when ready to test, I press the copy code "run button",
After the copying process is finished, usually type "import <lib. name>" to run my code in the terminal window.
I usually do not name the code main.py during dev cycles.

In order not to need to type that all the time, I would like to have some "buttons" inside the toolbar of the 
micropython-tools window, which can paste simple keystroke sequences and the Enter key into the "terminal window". 

Thank you for considering this!

ThomasAtBBTF, 2025-07-30:
> You can already paste text to the console the way things are if you have it in your clipboard.
Sure, this will work...
But I usually have "other things" in the clipboard from coding in the editor...

About "right-clicking", do you suggest doing the right click here?

[image]

This is not a real alternative!
- First, you would need to scroll the source file name into view,
- Then you would need to click there,
- Then you would need to move the mouse all the way down to the "Execute File in Repl" menu item and
- Then you would need to left click here.....

In this time, I can easily type "import <module-name>" and press Enter.

I see that in the menu there is the possibility to press shortcut keys...
This, I think, would also be a good alternative for my suggested "paste text" feature.
But this hotkey should use the filename of the file in the focused editor and not the filename of the selected tree item.

I just have the impression that the normal "Edit -> Save -> Run" user expericence would greatly benefit!

I would be as happy (or maybe even more happy) if a short cut key combination directly inside the editor "would do the trick"!

You can not believe how much this "IDE":

[image]

Changed the live of many programmers!
"Edit Compile Run Save" ....
With Hotkeys !

ThomasAtBBTF, 2025-07-30:
Yes, this is pretty much what I want! 
Thank you for pointing that out!

An additional "save and run" menu item in the context menu would be nice!

lukaskremla, 2025-07-30:
> Yes, this is pretty much what I want! Thank you for pointing that out!
> 
> An additional "save and run" menu item in the context menu would be nice!

Save and run - do you mean upload to device and run?

If so that seems like a rare use case for an action that would be displayed in a context menu - I reckon that would be better done with the before launch task run configuration chaining I suggested above.

ThomasAtBBTF, 2025-07-30:
Ok, fine with me!
With your suggestions and explanations, I am basically close enough to what I want.