lukaskremla/micropython-tools-jetbrains #57: Add support for editing on-device files

lukaskremla, 2025-07-30:


lukaskremla, 2025-08-02:
@sfe-SparkFro

I have a pre-release version of the edit on-device files feature available. It was too large for GitHub due to the bundled stub packages, so I had to host it on Proton:

https://drive.proton.me/urls/NVPRM97QZC#hkfx39o8KfRE

Opening of files is more polished now, they are tracked by the plugin, you can open a file with a certain remote path only once, the editor will redirect you to the already open file otherwise. 

I've also added an option to refresh the file, this is available both during and outside of editing. And of course I've added the option to edit the file itself, and to discard the changes. In edit mode both the refresh and cancel will display a warning dialog if you press them with unsaved changes.

Now - tracking open files perfectly per-device is impossible when working with MicroPython like so, the files are tracked through the remote paths.

So you can have two different files with identical paths on board1 and board2 and the plugin won't discriminate between them. If you open the file from board 1, disconnect it, and then connect to board2 and click refresh, the old file will be overwritten by the one on board2, if you click save, the file on board 2 will be overwritten by the one from board1.

Now that I'm writing this - maybe the save button should have a confirmation dialog too.

sfe-SparkFro, 2025-08-04:
Awesome! I gave it a quick test and works nicely, thank you!

2 minor things I noticed:

1. Would be nice if `Ctrl+S` would save the file, I know people love keyboard shortcuts!
2. MicroPython Tools doesn't detect if the board has been unplugged and replugged, so trying to save the file (or do anything, really) fails unless you manually click the disconnect and reconnect buttons. Would be great if it would at least automatically disconnected when the board is unplugged (bonus points for auto reconnecting!) though maybe that's a topic for a different issue.

lukaskremla, 2025-08-08:
1. `CTRL + S` shortcut makes sense, will add it - just need to verify it won't interfere with any existing IntelliJ functionality.
2. The plugin actually has connection integrity monitoring. However, it seems to not work on windows, or only work sometimes on windows... I'll see what's up with that.

lukaskremla, 2025-08-12:
https://drive.proton.me/urls/0AXYD34B7M#HkcQG4MxjFt0

Latest pre-release4 URL.

When editing an on-device file the plugin now listens for save shortcuts (cmd/ctrl + S by default). 

Connection integrity (plugin detecting the device's cable being yanked) should now work on windows too - I could only test with a Parallels VM, hopefully it's fixed for all Windows machines.

sfe-SparkFro, 2025-08-22:
Excellent, thank you!