lukaskremla/micropython-tools-jetbrains #62: Upload fails with "Uploaded files don't match with the expected sizes. Please try to re-run the upload."

Wout-VDWG, 2025-08-03:
Target device: ESP32-S3
Dev system: MacOS 14 - Apple M1 Pro - PyCharm 2025.1

I wrote a small Hello World test script as a first test to deploy on my ESP32-S3
I successfully installed the plugin, configured the settings and made a run config.

When I start the config, the upload preview dialog pops up highlighting my hello world script, I click "OK"
A upload progress bar can be seen for a few seconds, seemingly uploading something to the device.
Afterward, the progress bar closes, and an error shows at the bottom right inside PyCharm reading:

> Uploaded files don't match with the expected sizes. Please try to re-run the upload.

Not sure what I am doing wrong. Any help is welcome.

lukaskremla, 2025-08-04:
Could you please record a video of this? Ideally also showing the contents of the file you're uploading - and then opening the uploaded file to check it

lukaskremla, 2025-08-11:
@Wout-VDWG Are you still experiencing this issue? Did you have any time to check my previous comment?

lukaskremla, 2025-08-22:
I haven't been able to replicate this problem and no other issues were open about it.

If the problem persists for you please open a new issue.