lukaskremla/micropython-tools-jetbrains #68: Update

lukaskremla, 2025-08-28:
Update docs and readme for 0.6.0

MasterYuan418, 2025-08-28:
I'm now working on README and documentation for zh-CN(Simplified Chinese).

FYI: My fork[fork](https://github.com/MasterYuan418/micropython-tools-jetbrains)

lukaskremla, 2025-08-28:
Thank you! 

I'd suggest holding off on the readme and documentation. 

The changes aren't yet there, you'll save yourself some work. The bundle is in a mostly final stage.

I don't intend on changing any of the strings unless typos are discovered or the functionality changes significantly.

I'll update the readme and docs soon.

lukaskremla, 2025-09-02:
@MasterYuan418 README.md and DOCUMENTATION.md have been updated, they're up-to date in the main branch. The MpyBundle with the plugin's dialog text is also in its final form for now.